technototes/SkyStone2019 #9: We don't understand the Servo API at all

kevinfrei, 2019-11-08:
We aren't using the Servo API effectively. We need to have a better understanding of how things like "SetRange" work.

kevinfrei, 2019-11-08:
Docs: https://ftctechnh.github.io/ftc_app/doc/javadoc/index.html

kevinfrei, 2019-11-09:
Manual for the servo programmer:
http://www.revrobotics.com/content/docs/REV-31-1108-UM.pdf

ryantiotuico, 2019-11-09:
The servo was set to continuous rotation mode so the servo position API was only setting power and not position. A set position of 0.5 resulted in no movement. Position of 0 resulted in max speed in one direction. Position of 1 was max speed in opposite direction. 
The scaleRange API limited the speed of the servo. For example, when set to range of 0.4 to 0.6, the servo moved very slowly. The setDirection API changed from counter-clockwise to clockwise.

ryantiotuico, 2019-11-09:
In servo mode, setPosition moved it from left to right. scaleRange just limited to a smaller range of that. setDirection wasn't tested. Took 0.8 seconds for a 180 degree rotation.

closed 2019-11-09 by ryantiotuico